kbiggs/final-project-assignment-kbiggs #11: Implement application to run the HTML web server

mamo6538, 2024-04-03:


kbiggs, 2024-04-28:
Application running on RPi 3:
- Command line output showing connection to Tornado server and interactions for getting real-time data and updating settings

[image]

- Screenshot showing HTML page with real-time data for pressure sensors 

[image]

mamo6538, 2024-04-28:
The startup page:

[image]

Testing every button and receiving data:

[image]

Testing Alarms and settings:

[image]

closed 2024-04-28 by mamo6538